siara-cc/esp32_arduino_sqlite3_lib #1: Support for esp-idf

loboris, 2018-10-12:
There should be no problems to use this library with esp-idf (without Arduino-core).
It would be nice if you can include some examples of how to use it in a pure esp-idf project.

siara-cc, 2018-10-12:
sure.. i will check it out.. thanks..

siara-cc, 2018-10-17:
Please see https://github.com/siara-cc/esp-idf.

loboris, 2018-10-17:
Thanks, great job.
I'll test it and probably include it in my [MicroPython](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo) port in a next couple of weeks.

xpete, 2018-12-20:
It looks like the esp-idf project is not touched since Oct 17, 2018. Can you merge both project? Or even better, merge the 3 project to make the changes available on the 3 platforms at the same time.
Can you also change the esp-idf integration to work outside the eps-idf folder like any other third party component? Example: https://github.com/Molorius/esp32-websocket

siara-cc, 2018-12-20:
Thanks.. I will see how it goes..

siara-cc, 2018-12-31:
@xpete 
1. Could not merge code-set for `esp8266` and `esp32` because the version of `sqlite3` is different owing to memory limitation of `esp8266`.
2. Could not merge code-set for `esp32-arduino` and `esp32-idf` due to different folder structure.
3. I have made it similar to the 3rd party component you have mentioned and moved examples to separate repository [here](https://github.com/siara-cc/esp32-idf-sqlite3-examples).

I will keep up updates to the 3 repositories.

unangwarsodi, 2019-03-14:
SQL error: file is not a database
why?, i am using SDMMC on esp32
[image]

MY Setup:
Serial.begin(115200);
   sqlite3 *db1;
   sqlite3 *db2;
   char *zErrMsg = 0;
   int rc;

   SPI.begin();
   SD_MMC.begin();

   
   // list SPIFFS contents
   File root = SD_MMC.open("/");
   if (!root) {
       Serial.println("- failed to open directory");
       return;
   }
   if (!root.isDirectory()) {
       Serial.println(" - not a directory");
       return;
   }
   File file = root.openNextFile();
   while (file) {
       if (file.isDirectory()) {
           Serial.print("  DIR : ");
           Serial.println(file.name());
       } else {
           Serial.print("  FILE: ");
           Serial.print(file.name());
           Serial.print("\tSIZE: ");
           Serial.println(file.size());
       }
       file = root.openNextFile();
   }

   // remove existing file
//   SD_MMC.remove("/test1.db");
//   SD_MMC.remove("/test2.db");

   sqlite3_initialize();

   if (openDb("/sdcard/test1.db", &db1))
       return;
   if (openDb("/sdcard/test2.db", &db2))
       return;

//   rc = db_exec(db1, "CREATE TABLE test1 (id INTEGER, content);");
//   if (rc != SQLITE_OK) {
//       sqlite3_close(db1);
//       sqlite3_close(db2);
//       return;
//   }
//   rc = db_exec(db2, "CREATE TABLE test2 (id INTEGER, content);");
//   if (rc != SQLITE_OK) {
//       sqlite3_close(db1);
//       sqlite3_close(db2);
//       return;
//   }

   rc = db_exec(db1, "INSERT INTO test1 VALUES (1, 'Hello, World from test1');");
   if (rc != SQLITE_OK) {
       sqlite3_close(db1);
       sqlite3_close(db2);
       return;
   }
   rc = db_exec(db2, "INSERT INTO test2 VALUES (1, 'Hello, World from test2');");
   if (rc != SQLITE_OK) {
       sqlite3_close(db1);
       sqlite3_close(db2);
       return;
   }
   
   sqlite3_close(db1);
   sqlite3_close(db2);

tobozo, 2019-03-14:
I've had this issue too after stuffing a database with more fields than it could handle, for some reason the problem vanished after I "fixed" the micro SD card by having it repaired by windows.

Not sure if the FS driver uses some hacky methods to reallocate space, or if the maximum amount of writes is exceded in the FAT, but a full power cycle after each important operation (e.g. deleting db file, creating a database) often solved.

Also you may find helpful to always opendb() before every operation and sqlite3_close() after, whatever happens, don't leave any connection open.

unangwarsodi, 2019-03-14:
mean to my problem, the problem is with micro sd not in the script or library?

siara-cc, 2019-03-14:
Hi @unangwarsodi,  I am not sure why but what @tobozo has said may be the reason (corrupted Micro SD card because of power fluctuation).  In addition to what he has suggested, please refer to the issue I fixed recently:
https://github.com/siara-cc/esp32-idf-sqlite3/issues/2#issue-416938715
If you are using a older version, please use the latest version.  But if it is already corrupted, this will not fix it.  If you are able to repeat it consistently with a formatted card, please let me know the steps so I can troubleshoot. Thanks.

unangwarsodi, 2019-03-15:
I tried to reformat my Micro SD, after I tried again and it worked. thank you

unangwarsodi, 2019-03-15:
STRING OR BLOOB To BIG
query1: rc = db_exec(db1, "CREATE TABLE tbl_template2 (id INTEGER PRIMARY KEY, finger_id INTEGER, device_id INTEGER, template_finger);");

query2: rc = db_exec(db1, "CREATE TABLE tbl_template2 (id INTEGER PRIMARY KEY, finger_id INTEGER, device_id INTEGER, template_finger TEXT);");

I try two query, but the same. query2 max length 225 more than that will issue the error. and for query1 I tried to store data with a length of more than 255 and it worked, but then trying with the length of data 1237 gave rise to an error message STRING OR BLOOB To BIG, my data length is approximately 5000 characters. what data type is suitable for storing data with characters longer than 5000 in sqlite?, I tried with TEXT data types but it didn't work

siara-cc, 2019-03-15:
Please see new issue created for tracking.  I am closing this.

closed 2019-03-15 by siara-cc