siara-cc/esp32_arduino_sqlite3_lib #2: ESP SDK 1.0.1-rc

tobozo, 2018-11-28:
Hello and thanks for this awesome library !

I've been playing with it on ESP32 for a few weeks now and it [works like a charm](https://github.com/tobozo/ESP32-BLECollector).

Last night Espressif did a release candidate for the 1.0.1 SDK for Arduino IDE (which you can update from your boards manager) and today the sqlite3 library won't compile.

<details>
  <summary> Compile log (...)</summary>

```
In file included from ~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:13476:0:
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c: In function 'pcache1TruncateUnsafe':
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:48838:11: error: 'nPage' undeclared (first use in this function)
   assert( nPage<0 || pCache->nPage==(unsigned)nPage );
           ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                    ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:48838:11: note: each undeclared identifier is reported only once for each function it appears in
   assert( nPage<0 || pCache->nPage==(unsigned)nPage );
           ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c: In function 'pcache1Fetch':
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:49201:11: error: 'pCache' undeclared (first use in this function)
   assert( pCache->bPurgeable || createFlag!=1 );
           ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c: In function 'sqlite3PagerUnrefNotNull':
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:55770:33: error: 'pPager' undeclared (first use in this function)
   assert( sqlite3PcacheRefCount(pPager->pPCache)>0 );
                                 ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c: In function 'cellSizePtr':
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:63969:18: error: 'debuginfo' undeclared (first use in this function)
   assert( nSize==debuginfo.nSize || CORRUPT_DB );
                  ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c: In function 'cellSizePtrNoPayload':
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:63990:11: error: 'debuginfo' undeclared (first use in this function)
   assert( debuginfo.nSize==(u16)(pIter - pCell) || CORRUPT_DB );
           ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c: In function 'balance':
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:70758:17: error: 'balance_deeper_called' undeclared (first use in this function)
         assert( balance_deeper_called==0 );
                 ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c: In function 'sqlite3_backup_step':
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:73439:15: error: 'rc2' undeclared (first use in this function)
       assert( rc2==SQLITE_OK );
               ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c: In function 'sqlite3Step':
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:81073:64: error: 'Vdbe {aka struct Vdbe}' has no member named 'rcApp'
   assert( (p->rc!=SQLITE_ROW && p->rc!=SQLITE_DONE) || p->rc==p->rcApp );
                                                                ^
[... 134 lines cut ...]
               ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:106230:11: error: 'nLookaside' undeclared (first use in this function)
   assert( nLookaside==0 || nLookaside==sqlite3LookasideUsed(db,0) );
           ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c: In function 'sqlite3InitCallback':
~\Arduino\libraries\esp32_arduino_sqlite3_lib\src\sqlite3.c:122300:25: error: 'rcp' undeclared (first use in this function)
     assert( (rc&0xFF)==(rcp&0xFF) );
                         ^
~\ArduinoData\packages\esp32\hardware\esp32\1.0.1-rc1/tools/sdk/include/newlib/assert.h:14:37: note: in definition of macro 'assert'
 # define assert(__e) ((void) sizeof(__e))
                                     ^
```

</details>
<br><br>

Since I can easily rollback to the 1.0.0 and get a fully functional version this isn't a blocker for me, just leaving this issue here so anyone in the same situation can followup.

I'll be posting an issue on espressif tracker too.

siara-cc, 2018-11-28:
I will check what the problem is. Please report on espressif later unless you are facing similar issue elsewhere.

tobozo, 2018-11-28:
[ongoing discussion](https://gitter.im/espressif/arduino-esp32?at=5bfe6d92e25cc27405572d67)

tobozo, 2018-11-29:
I did this after reading one of your comments on the gitter : 
[image]

And it compiles fine in the Arduino IDE indeed, it even shows me a 5% dynamic memory improvement and while I'm not sure where this comes from, it's still welcome :-)

siara-cc, 2018-11-29:
Just waiting to see what @igrr would say.. Changing sqlite3.c is always possible.

BTW, I am not sure why you should face memory issues with this library.  It is supposed to use 64k memory max for caching according to my understanding, unless you are using a large page size.  Even if more allowances are made I think it should not crash. Of course this is not related to the issue on hand.

tobozo, 2018-11-29:
I probably face memory issues as a punishment for the all the bad practices in my code :]

siara-cc, 2018-12-20:
Sorry I saw your edit above just now... Since there is no reply from ESP, I just fixed the issue by changing sqlite3.c.  If you still face issue, please let me know.

siara-cc, 2018-12-20:
It appears github does not notify for "edits".

tobozo, 2018-12-20:
OMG github managed to detect and display the diff in the middle of this huge file, this is worth a screenshot :D 
[image]

I'll give it a shot right now.

tobozo, 2018-12-20:
Looks fine with the esp32-arduino rc2 so I can close this issue \o/ 
Thanks for the followup!

In another topic I'll apoligize in advance fort the unsollicited advice, but you seem to be struggling with your tweeter account, kinda deleted the only tweet that was actually buzzing and posted the same tweet multiple times after that, so here are a few unsollicited tips:

- The more redundant the contents, the more penalized the account. Make sure all your tweets are different, or group all the recipients
- It's okay to retweet yourself or answer to your own tweet as a followup
- Make sure you follow at least 200 accounts before tweeting an announcement (following them is enough to eventually reach them), and follow at least 10 more people every day until you reach critical mass
- Follow back whoever relevant account follows you
- Mute the irrelevant accounts who follow you (i.e. fake news bots) so they don't not pollute your follower's timelines (and yours)

closed 2018-12-20 by tobozo

siara-cc, 2018-12-20:
Thanks.. I will make a release soon, so you can have a reference.

You are right.. I am struggling with Twitter.. this is the best advice I have received recently.. Thanks - I highly appreciate it.