siara-cc/esp32_arduino_sqlite3_lib #3: Arduino Library Manager

tobozo, 2019-02-15:
hope you don't mind I made the request for you ^_^

siara-cc, 2019-02-15:
Wow.. thanks @tobozo .. I didn't know about it.. I hope many people use it..

tobozo, 2019-03-01:
looks like it's passed the tests successfully \o/ it should land in the Library Manager in a couple of hours

tobozo, 2019-03-01:
funny the platformio robot didn't find your project though, I've given it a little help:

```
$ pio lib register https://github.com/siara-cc/esp32_arduino_sqlite3_lib/raw/master/library.properties
The library has been successfully registered and is waiting for moderation
```

closed 2019-03-01 by tobozo

siara-cc, 2019-03-01:
Many thanks @tobozo.. Can I request to add myself or only users can?

tobozo, 2019-03-01:
I'm not a power user, anyone can request it

siara-cc, 2019-03-01:
Yes!!  It is on their website.  But they seem to have taken the oldest release though!!  Anyway, the link is there for anyone to upgrade.
[image]

tobozo, 2019-03-01:
that's another registry, platformio registry is that one: https://platformio.org/lib/search?query=sqlite3

actually I don't even know if this library works on platformio, but it's only a matter of adding a plaformio.ini with the correct -D flags  (and the corresponding #ifdef's in the code)

siara-cc, 2019-03-01:
yes, I have heard about PIO.. I haven't used it either..

tobozo, 2019-03-01:
well I just tried to import it via `pio lib install https://github.com/siara-cc/esp32_arduino_sqlite3_lib.git` and it compiled the spiffs example just fine

tobozo, 2019-03-01:
congrats !!!

[image]

tobozo, 2019-03-01:
this will only make your *releases* available from the Library Manager, totally different from doing a git clone from the master

siara-cc, 2019-03-01:
right.. GUI Frontend that will cover non-CLI users too.

tobozo, 2019-03-01:
Actually the current release does not compile on Arduino IDE (I had to overwrite the library folder with the current master source) so you may want to increment your version in library.properties, then create a tag and produce a release.

Platformio and Arduino library managers will automatically detect this release and send and add it to the versions list (no need to re-submit) as long as the library.properties version and release values match, and the version increments.

tobozo, 2019-03-02:
And platformio coped with the new version https://platformio.org/lib/show/6173/Sqlite3Esp32 \o/

siara-cc, 2019-03-02:
Excellent.. I also requested to add my other libraries on both platforms.. Thanks @tobozo ...