siara-cc/esp32_arduino_sqlite3_lib #8: database-disk-image-malformed

unangwarsodi, 2019-03-18:
can one database have many tables?, I try one database of two tables and it doesn't work and raises a "database-disk-image-malformed" error message in the second table

siara-cc, 2019-03-18:
One database can have many tables.  I may be that your sd card is getting corrupted often presumably because of power fluctuations.   I suggest you to try storing the database on the internal SPIFFS.  It could have more tolerance to such issues.

Also, please note these storage media (sdcard or spiffs) are not like hard disk or RAM.  They have limitation on how many writes can be made to each sector.  Usually after more than 10000 writes to one sector it gives such problems.  So to avoid such problems, need to ensure that same area is not written very often.

siara-cc, 2019-03-18:
https://github.com/siara-cc/esp32_arduino_sqlite3_lib#limitations-of-flash-memory

unangwarsodi, 2019-03-18:
can you give an example of the script?
Thanks

siara-cc, 2019-03-19:
For using SPIFFS, there is already an example in the repo.

unangwarsodi, 2019-03-19:
how to reopen connection database after close?

closed 2019-03-19 by siara-cc