siara-cc/esp32_arduino_sqlite3_lib #10: SQL error: SQL logic error when try to rename table name

pornpol, 2019-03-22:
SQL error: SQL logic error when I try to rename a table.

> ALTER TABLE aa RENAME TO bb;
> SQL error: SQL logic error

siara-cc, 2019-03-22:
I am unable to immediately ascertain whether this feature enabled or not.
Is it really needed on a embedded device?

pornpol, 2019-03-22:
Because of I cannot insert more data in one table due to memory limit.
Then I try to rename old table and create new table to insert new data.

siara-cc, 2019-03-22:
I see.. is it possible to keep the original name and give new name for the new table?  It may take me sometime to get RENAME working.  It was recently introduced and I am even thinking to rollback to older version SQLite because of memory limitations, if people are not using the new features.

tobozo, 2019-03-22:
@pornpol you'll have less problems if you don't rename *after* creating the table and instead generate a unique name (timestamp?) *before* creating any table + file

pornpol, 2019-03-28:
I create new table with timestamp and no nned to rename anymore. Thank you.

closed 2019-03-28 by pornpol

ANSB, 2021-07-23:
If you want to really rename a table here is a way to do that: 
https://arduino.stackexchange.com/questions/85280/esp32-and-sqlite-alter-table-problem/85297#85297