siara-cc/esp32_arduino_sqlite3_lib #11: ESP32 with Arduino Framework Freezes

erenay93, 2019-03-29:
Hi,
I have such algorithm that request SELECT query every second on ESP32 and sql file on SPIFFS. It works well but at a random time it throws an exception and ESP32 just freezes. I have searched and found that the error is related with accessing to flash memory while ISR triggered. But I can't resolve the problem.

Here is Console output after problem occured.

`15:28:51.228 -> Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
15:28:51.228 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
15:28:51.228 -> Memory dump at 0x40128b24: bad00bad bad00bad bad00bad
15:28:51.228 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
15:28:51.228 -> Memory dump at 0x40128b24: bad00bad bad00bad bad00bad
15:28:51.262 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
15:28:51.262 -> Memory dump at 0x40128b24: bad00bad bad00bad bad00bad
15:28:51.262 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
15:28:51.262 -> Memory dump at 0x40128b24: bad00bad bad00bad bad00bad
15:28:51.262 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
15:28:51.300 -> Memory dump at 0x40128b24: bad00bad bad00bad bad00bad
15:28:51.300 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
15:28:51.300 -> Memory dump at 0x40128b24: bad00bad bad00bad bad00bad
15:28:51.300 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
15:28:51.300 -> Memory dump at 0x40128b24: bad00bad bad00bad bad00bad
15:28:51.300 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
15:28:51.349 -> Memory dump at 0x40128b24: bad00bad bad00bad bad00bad
15:28:51.349 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
15:28:51.349 -> Memory dump at 0x40128b24: bad00bad bad00bad bad00bad
15:28:51.349 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
15:28:51.349 -> Memory dump at 0x40128b24: bad00bad bad00bad bad00bad
15:28:51.349 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
15:28:51.349 -> Memory dump at 0x40128b24: bad00bad bad00bad bad00bad
15:28:51.349 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
15:28:51.386 -> Memory dump at 0x40128b24: bad00bad bad00bad bad00bad
15:28:51.386 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
15:28:51.386 -> Memory dump at 0x40128b24: bad00bad bad00bad bad00bad
15:28:51.386 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
15:28:51.386 -> Memory dump at 0x40128b24: bad00bad bad00bad bad00bad
15:28:51.386 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
15:28:51.433 -> Memory dump at 0x40128b24: bad00bad bad00bad bad00bad
15:28:51.433 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
15:28:51.433 -> Memory dump at 0x40128b24: bad00bad bad00bad bad00bad
15:28:51.433 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
15:28:51.433 -> Memory dump at 0x40128b24: bad00bad bad00bad bad00bad
15:28:51.433 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
15:28:51.433 -> Memory dump at 0x40128b24: bad00bad bad00bad bad00bad
15:28:51.433 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
15:28:51.480 -> Memory dump at 0x40128b24: bad00bad bad00bad bad00bad
15:28:51.480 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
15:28:51.480 -> Memory dump at 0x40128b24: bad00bad bad00bad bad00bad
15:28:51.480 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
15:28:51.480 -> Memory dump at 0x40128b24: bad00bad bad00bad bad00bad
15:28:51.480 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
15:28:51.480 -> Memory dump at 0x40128b24: bad00bad bad00bad bad00bad
15:28:51.518 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
15:28:51.518 -> Memory dump at 0x40128b24: bad00bad bad00bad bad00bad
15:28:51.518 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
15:28:51.518 -> Memory dump at 0x40128b24:Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
15:28:51.518 -> Debug exception reason: BREAK instr 
Guru Meditation Error: Core  1 panic'ed (Double exception)
15:28:51.565 -> Guru Meditation Error: Core  1 panic'ed (Double exception)
15:28:51.565 -> Guru Meditation Error: Core  1 panic'ed (`

ESP32 stoped working and even watchdog not triggered.

If its required I may share the code.

Thanks for your help.

siara-cc, 2019-03-29:
Hi, Thank you for using this library.  I have heard about the problem you are facing:
https://github.com/micropython/micropython/issues/3782
It will take me sometime to recreate the problem and see what can be done.  Meanwhile there are two approaches:
- Copy the DB into memory if it will fit. Please see: https://sqlite.org/inmemorydb.html
- Since the stalls are sub-second, there still might be room for you to overcome it - but as of now I am not sure how.

siara-cc, 2019-03-29:
You might also think about using PSRAM, which is additional RAM (~4GB) on some boards which might allow your DB be kept in memory.  These are not too expensive.

erenay93, 2019-04-01:
Thank you very much for your reply.
I need to store that data until send them over internet even if ESP32 restart.
Do you have any idea about that situation.

siara-cc, 2019-04-01:
From your first question, I thought you are trying to SELECT, but now it appears you are trying to store.. If you are facing issues with storing to DB, please try writing to a regular file.  From reading https://github.com/micropython/micropython/issues/3782, it appears it only freezes and does not crash or restart. So I guess it should work for you.

erenay93, 2019-04-01:
Sorry for miss information, that my fault.
I have INSERT and SELECT queries but I'm facing problem with SELECT queries.

erenay93, 2019-04-02:
By the way I've tried that library with SD card and it works perfect for 5 hours. 2 INSERT and 2 SELECT queries are executing per second for testing purposes and Its working very well. And for further information, I've tried an other DB library called [Arduino Extended Database Library](https://github.com/jwhiddon/EDB), It has similar problem with the onboard flash memory. So I think there is a problem with flash memory read/write on ESP32 core.

siara-cc, 2019-04-02:
@erenay93 Thank you for the information.  It may be a good idea to raise it on ESP github site.  They may come up with a solution.

siara-cc, 2019-04-04:
@erenay93 Please see the conversation at this forum at around this time: https://gitter.im/espressif/arduino-esp32. There is another user who is facing similar crash.. The user `atanisoft` shows how to figure out what is wrong by using the stack trace decoder (If you have the `Backtrace:` dump from your crash).

It may be possible to overcome your problem with writing to SPIFFS.

siara-cc, 2019-04-14:
@erenay93 I am closing this issue as it is a problem of accessing SPIFFS through ISR.  If you think otherwise, please create new issue or append here.  I actually tested using the bulk insert example appending 1000 records on SPIFFS.  It does become slow and there is a latency of 2 to 3 seconds per insert on and off.

By the way, there are some significant changes related to stability.  Please try by cloning the repo if you have time. Thanks.

closed 2019-04-14 by siara-cc