siara-cc/esp32_arduino_sqlite3_lib #13: Disk I/O issues on SPIFFS

txf-, 2019-07-19:
Currently I'm trying to insert 1000+ entries into a db on SPIFFS with a  modified sqlite_bulk_data_insert.ino, 

I'm getting periodic disk I/O errors, seemingly at random, but with increasing frequency the larger the database becomes. Notably executing further inserts in the db presents no issues. 

Looking at the heap values, they remain unchanged, could it be that the heap is getting fragmented?

siara-cc, 2019-07-19:
I remember testing on SPIFFS when I published that example. But I don't remember what the outcome was. I will try it out again and get back to you.

siara-cc, 2019-08-23:
I tried the example on SPIFFS, but I did not face the problems you have mentioned.  I could not complete the test because of hardware failure.

About heap values, my opinion is that they plateau once there are no more new sized data need to be allocated.  They seem not to change because the same sized data gets allocated and deallocated.

siara-cc, 2019-11-17:
See new library https://github.com/siara-cc/sqlite_micro_logger_arduino.
It overcomes all memory problems and corruption issues.
However there are limitations.
For documentation see: https://github.com/siara-cc/sqlite_micro_logger_c.

closed 2020-03-08 by siara-cc