siara-cc/esp32_arduino_sqlite3_lib #16: Multiple rows in a single query

vanBassum, 2019-11-27:
Hello,

I am trying to add multiple rows in a single quary. This is however not working as expected. When I try this, only the last 'row' gets added to the table. I have tried 2 different quaries:

`REPLACE INTO 'Settings' SELECT '1' AS 'KCCMD', '0' AS 'Value'UNION SELECT 'ATS', '20.000000'UNION SELECT 'BOR', '0
'UNION SELECT 'BOS', '0'UNION SELECT 'CCA', '0'UNION SELECT 'CCU', '0'UNION SELECT 'CHA', '0'UNION SELECT 'CHU', '0'UNION SELECT 'DCC'
, '1.000000'UNION SELECT 'DHC', '1.000000'UNION SELECT 'DTN', '0.000000'UNION SELECT 'ETS', '16.000000'UNION SELECT 'MIT', '15.000000'
UNION SELECT 'MXT', '24.000000'UNION SELECT 'SCC', 'true'UNION SELECT 'THR', '0'UNION SELECT 'THS', '0'`

`REPLACE INTO 'Settings' (KCCMD, Value) VALUES('ATS', '20.000000'),('BOR', '0'),('BOS', '0'),('CCA', '0'),('CCU', '
0'),('CHA', '0'),('CHU', '0'),('DCC', '1.000000'),('DHC', '1.000000'),('DTN', '0.000000'),('ETS', '16.000000'),('MIT', '15.000000'),('
MXT', '24.000000'),('SCC', 'true'),('THR', '0'),('THS', '0');
`

siara-cc, 2019-11-28:
Hi, Thank you for reaching out.  I will check - I work on these projects only during weekends. Does this work on the desktop?

vanBassum, 2019-11-28:
Yes, both queries work with "https://sqlitebrowser.org/".
I downloaded the database from the ESP that was created with this lib and opened in the sqlitebrowser to test this. I have also tested the queries with INSERT instead of REPLACE, still no luck.

namoaton, 2020-01-24:
can  perform INSERT query?

siara-cc, 2020-03-08:
@vanBassum I tried your statements are working, but only when I use a smaller set of data, probably because sqlite_console example does not have a big enough buffer.  The second replace worked as it is, but the first one restarted esp32.
Try cloning the latest version of this repo, because could also be because I upgraded version of Sqlite.

closed 2020-03-08 by siara-cc