siara-cc/esp32_arduino_sqlite3_lib #17: Unable to Compile Example Sketch on Arduino IDE

fmanzoor, 2019-11-27:
I am having problem compiling my sketch and one of the example sketch that came with Arduino IDE's downloaded library. Shouldn't the code (at least the example) compile out of box?

```
Arduino: 1.8.10 (Linux), Board: "FireBeetle-ESP32, 80MHz, 115200"

/home/fahad/Downloads/arduino-1.8.10/arduino-builder -dump-prefs -logger=machine -hardware /home/fahad/Downloads/arduino-1.8.10/hardware -hardware /home/fahad/.arduino15/packages -tools /home/fahad/Downloads/arduino-1.8.10/tools-builder -tools /home/fahad/Downloads/arduino-1.8.10/hardware/tools/avr -tools /home/fahad/.arduino15/packages -built-in-libraries /home/fahad/Downloads/arduino-1.8.10/libraries -libraries /home/fahad/Arduino/libraries -fqbn=esp32:esp32:firebeetle32:FlashFreq=80,UploadSpeed=115200 -ide-version=10810 -build-path /tmp/arduino_build_596537 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/home/fahad/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=/home/fahad/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=/home/fahad/.arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=/home/fahad/.arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.mkspiffs.path=/home/fahad/.arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/home/fahad/.arduino15/packages/esp32/tools/mkspiffs/0.2.3 -verbose /home/fahad/Arduino/libraries/Sqlite3Esp32/examples/sqlite3_spiffs/sqlite3_spiffs.ino
/home/fahad/Downloads/arduino-1.8.10/arduino-builder -compile -logger=machine -hardware /home/fahad/Downloads/arduino-1.8.10/hardware -hardware /home/fahad/.arduino15/packages -tools /home/fahad/Downloads/arduino-1.8.10/tools-builder -tools /home/fahad/Downloads/arduino-1.8.10/hardware/tools/avr -tools /home/fahad/.arduino15/packages -built-in-libraries /home/fahad/Downloads/arduino-1.8.10/libraries -libraries /home/fahad/Arduino/libraries -fqbn=esp32:esp32:firebeetle32:FlashFreq=80,UploadSpeed=115200 -ide-version=10810 -build-path /tmp/arduino_build_596537 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/home/fahad/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=/home/fahad/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=/home/fahad/.arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=/home/fahad/.arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.mkspiffs.path=/home/fahad/.arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/home/fahad/.arduino15/packages/esp32/tools/mkspiffs/0.2.3 -verbose /home/fahad/Arduino/libraries/Sqlite3Esp32/examples/sqlite3_spiffs/sqlite3_spiffs.ino
Using board 'firebeetle32' from platform in folder: /home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3
Using core 'esp32' from platform in folder: /home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3
Detecting libraries used...
/home/fahad/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/config -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/app_trace -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/app_update -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/asio -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/bootloader_support -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/bt -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/coap -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/console -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/driver -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/esp-tls -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/esp32 -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/esp_adc_cal -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/esp_event -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/esp_http_client -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/esp_http_server -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/esp_https_ota -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/esp_ringbuf -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/ethernet -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/expat -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/fatfs -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/freemodbus -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/freertos -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/heap -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/idf_test -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/jsmn -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/json -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/libsodium -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/log -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/lwip -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/mbedtls -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/mdns -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/micro-ecc -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/mqtt -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/newlib -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/nghttp -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/nvs_flash -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/openssl -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/protobuf-c -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/protocomm -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/pthread -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/sdmmc -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/smartconfig_ack -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/soc -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/spi_flash -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/spiffs -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/tcp_transport -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/tcpip_adapter -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/ulp -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/vfs -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/wear_levelling -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/wifi_provisioning -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/wpa_supplicant -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/xtensa-debug-module -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/esp-face -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/esp32-camera -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/include/esp-face -I/home/fahad/.arduino15/packages/esp32/hardware/esp3 [... cut ...]
Using library Sqlite3Esp32 at version 2.0 in folder: /home/fahad/Arduino/libraries/Sqlite3Esp32 
Using library SPI at version 1.0 in folder: /home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/libraries/SPI 
Using library FS at version 1.0 in folder: /home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/libraries/FS 
Using library SPIFFS at version 1.0 in folder: /home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.3/libraries/SPIFFS 
exit status 1
Error compiling for board FireBeetle-ESP32.
```

fmanzoor, 2019-11-27:
Don't know if this is  an IDE related issue or something else but I was able to compile my own Sketch (did not test the example) using [Makefile](https://github.com/plerup/makeEspArduino/blob/master/makeEspArduino.mk) and there was no error/warning. Let me know if anything else is needed to diagnose the issue.

siara-cc, 2019-11-28:
Hi, Thank you for reaching out.  I will check - I work on these projects only during weekends.

ashna-zafar, 2019-12-21:
hello..can anyone help me please  i am beginner  in esp32......i am trying to compile my sketch.....but one error continuously occurs
Error compiling for board ESP32 Dev Module......i have tried many different things like i have installed core zip files  by follow this link https://www.esp8266.com/viewtopic.php?f=26&t=14222&start=4    
below is my sketch
#include <Arduino.h>
#include "BluetoothSerial.h"
#include <IRremote.h>
#include <IRac.h>
#include <IRrecv.h>
#include <IRsend.h>
#include <IRtimer.h>
#include <IRutils.h>
#include <ir_NEC.h>
#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

#define NEC_BITS           32
#define DUTY_CYCLE         1/3       // Carrier duty cycle
#define PERIOD             26        // 38kHz, 1/38kHz =~ 26us
#define ACTIVE_HIGH       (PERIOD * DUTY_CYCLE)
#define ACTIVE_LOW        (PERIOD - (PERIOD * DUTY_CYCLE)) 
#define Power 0xFBD22D 
#define Up 0xFBE817 
#define Down 0xFB38C7

#define TOPBIT 0x80000000
#define NEC_HDR_MARK      9000
#define NEC_HDR_SPACE     4500
#define NEC_BIT_MARK       560
#define NEC_ONE_SPACE     1690
#define NEC_ZERO_SPACE     560
char code;

BluetoothSerial SerialBT;
//int incoming;
char incomingChar;// received value will be stored as CHAR in this variable
const int OutputPin =13;

void setup() {
  Serial.begin(9600); //Start Serial monitor in 9600
  SerialBT.begin("ESP32test"); //Bluetooth device name
  Serial.println("The device started, now you can pair it with bluetooth!");
  Serial.println("To Power send: a");//print on serial monitor 
  Serial.println("To Up send: b");
  Serial.println("To Down send: c");
  pinMode(OutputPin, OUTPUT);  
}
void sendNEC(unsigned long data, int nbits) {
  mark(NEC_HDR_MARK);
  space(NEC_HDR_SPACE);

  for (int i = 0; i < nbits; i++) {
    if (data & TOPBIT) {
      mark(NEC_BIT_MARK);
      space(NEC_ONE_SPACE);
    }
    else {
      mark(NEC_BIT_MARK);
      space(NEC_ZERO_SPACE);
    }
    data <<= 1;
  }
  mark(NEC_BIT_MARK);
  space(NEC_HDR_SPACE);
  mark(NEC_BIT_MARK);
  space(0);
}
void mark(int time) {
  //inverting the signal
  digitalWrite(OutputPin, LOW);
  delayMicroseconds(time);     //digitalwrite removes the modulation to provide a demodulated signal
}
void space(int time) {
  digitalWrite(OutputPin, HIGH);
  delayMicroseconds(time);
}
void loop() {
    incomingChar =(char)SerialBT.read();
  if (Serial.available())
  {
    SerialBT.write(Serial.read());
  }
   if (SerialBT.available()) 
    {code = SerialBT.read(); //Read what we recevive 
     Serial.print("Received:"); 
                                //Serial.println(receivedChar);//print on serial monitor
    //Serial.println(code);
    //Serial.println(incoming);
    SerialBT.print ("Received:");//print on serial monitor
                                  //SerialBT.println(receivedChar);//print on serial monitor
   

     if(code == Power)
   {Serial.println("Power received");
    sendNEC(Power, 32);
    Serial.println(code);
    if(code == Up )
    {digitalWrite(OutputPin, HIGH);
     Serial.println("Up received");
     sendNEC(Up, 32);}
     if(code == Down)
     {digitalWrite(OutputPin, LOW);
      Serial.println("Down received");
      sendNEC(Down, 32);}
      }
  else
   {
    sendNEC(Power, 32);
   }
  delay(20);
}
}

siara-cc, 2019-12-22:
Hi @ashna-zafar, your sketch does not use this library. So closing as unrelated. Also, you have not mentioned what the error is.

@fmanzoor I tried compiling the sketch with Arduino IDE 1.8.5 and ESP32 core 1.0.4, it is compiling fine.  If you are still facing issue, please let me know with compilation log.

tobozo, 2019-12-22:
@siara-cc I suspect that a project name prefixed "esp32_arduino" can somehow be confused with the platform project. 

@ashna-zafar did you mean to post on this site ? https://github.com/espressif/arduino-esp32

fmanzoor, 2019-12-22:
@siara-cc the log file (of the sqlite3_spiffs example) has been shared earlier and the problem is persisting with the IDE but I am able to compile my own sketch using the Makefile that I referred earlier. Here is the freshest log file of the sketch compilation nonetheless and by the way, I am using IDE 1.8.10 and 1.0.4 version of the ESP32 core.

```
Arduino: 1.8.10 (Linux), Board: "FireBeetle-ESP32, 80MHz, 115200"

/home/fahad/Downloads/arduino-1.8.10/arduino-builder -dump-prefs -logger=machine -hardware /home/fahad/Downloads/arduino-1.8.10/hardware -hardware /home/fahad/.arduino15/packages -tools /home/fahad/Downloads/arduino-1.8.10/tools-builder -tools /home/fahad/Downloads/arduino-1.8.10/hardware/tools/avr -tools /home/fahad/.arduino15/packages -built-in-libraries /home/fahad/Downloads/arduino-1.8.10/libraries -libraries /home/fahad/Arduino/libraries -fqbn=esp32:esp32:firebeetle32:FlashFreq=80,UploadSpeed=115200 -vid-pid=10C4_EA60 -ide-version=10810 -build-path /tmp/arduino_build_968988 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=/home/fahad/.arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=/home/fahad/.arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/home/fahad/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=/home/fahad/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.mkspiffs.path=/home/fahad/.arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/home/fahad/.arduino15/packages/esp32/tools/mkspiffs/0.2.3 -verbose /home/fahad/Arduino/libraries/Sqlite3Esp32/examples/sqlite3_spiffs/sqlite3_spiffs.ino
/home/fahad/Downloads/arduino-1.8.10/arduino-builder -compile -logger=machine -hardware /home/fahad/Downloads/arduino-1.8.10/hardware -hardware /home/fahad/.arduino15/packages -tools /home/fahad/Downloads/arduino-1.8.10/tools-builder -tools /home/fahad/Downloads/arduino-1.8.10/hardware/tools/avr -tools /home/fahad/.arduino15/packages -built-in-libraries /home/fahad/Downloads/arduino-1.8.10/libraries -libraries /home/fahad/Arduino/libraries -fqbn=esp32:esp32:firebeetle32:FlashFreq=80,UploadSpeed=115200 -vid-pid=10C4_EA60 -ide-version=10810 -build-path /tmp/arduino_build_968988 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=/home/fahad/.arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=/home/fahad/.arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/home/fahad/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=/home/fahad/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.mkspiffs.path=/home/fahad/.arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/home/fahad/.arduino15/packages/esp32/tools/mkspiffs/0.2.3 -verbose /home/fahad/Arduino/libraries/Sqlite3Esp32/examples/sqlite3_spiffs/sqlite3_spiffs.ino
Using board 'firebeetle32' from platform in folder: /home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.4
Using core 'esp32' from platform in folder: /home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.4
Detecting libraries used...
/home/fahad/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/config -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/app_trace -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/app_update -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/asio -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/bootloader_support -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/bt -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/coap -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/console -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/driver -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp-tls -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp32 -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_adc_cal -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_event -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_http_client -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_http_server -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_https_ota -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_ringbuf -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/ethernet -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/expat -I/home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/inclu [... cut ...]
Using library Sqlite3Esp32 at version 2.0 in folder: /home/fahad/Arduino/libraries/Sqlite3Esp32 
Using library SPI at version 1.0 in folder: /home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SPI 
Using library FS at version 1.0 in folder: /home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/FS 
Using library SPIFFS at version 1.0 in folder: /home/fahad/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SPIFFS 
exit status 1
Error compiling for board FireBeetle-ESP32.
```

siara-cc, 2019-12-22:
@fmanzoor First time I am hearing a Makefile can be used to compile an Arduino sketch.  From the log, it looks like it is treating warnings as errors.  If you have changed the settings for the compiler somehow, you may try by reverting them.  Thats as much as I can deduce from the inputs.

fmanzoor, 2019-12-27:
I do not recall changing any settings or compiler/linker flags and I do not even know where/how these settings can be changed in Arduino IDE. Is there any particular file/path that I should check for modified settings that I might have erroneously made?

Do check out https://github.com/plerup/makeEspArduino as it makes life easy because Arduino IDE compiles complete project (including the ESP32 core) every time when you upload sketch but in case of the Makefile approach, the toolchain is applied only to the file where code is modified unless, you `make clean` before `make` command.

siara-cc, 2019-12-28:
@fmanzoor Not sure if the compiler settings for Arduino IDE can be changed.
I tried upgrading to 1.8.10 and `sqlite3_spiffs` compiles fine.  You could try using `ESP32-Dev-module` instead of `FireBeetle-ESP32`.  Or even use a fresh installation of 1.8.10 and `arduino-esp32` core as the `makeEspArduino` may have changed some settings.

tobozo, 2019-12-28:
@fmanzoor having `-warnings` set to `all` instead of `default` does this indeed

[image]

siara-cc, 2019-12-28:
Thanks @tobozo 
@fmanzoor Although this library exposes the Sqlite library functions as it is with SQL support etc, it struggles because of limited memory on ESP32.
I recommend my other library [Sqlite Micro Logger](https://github.com/siara-cc/sqlite_micro_logger_arduino) which supports most functions (but no SQL) using minimal memory usage, that it works even on Arduino Uno!!

fmanzoor, 2020-02-10:
@tobozo Your suggestion did work and compilation was failing due to my negligence but in my defense I thought that having those warnings (on) during the compilation phase would help in reducing any possible pointer-related issues during runtime.

closed 2020-02-10 by fmanzoor

tobozo, 2020-02-15:
@siara-cc @fmanzoor  found a cleaner way by adding this to `sqlite3.c`, it makes the warnings go silent for sqlite while keeping the initial flags for the rest of the code:

```C
    #pragma GCC diagnostic push
    #pragma GCC diagnostic ignored "-Wunused-value"
    #pragma GCC diagnostic ignored "-Wunused-label"
    #pragma GCC diagnostic ignored "-Wchar-subscripts"

    // (.....)

    #pragma GCC diagnostic pop
```

I also added this to `shox96_0_2.cpp`:

```C
    #pragma GCC diagnostic push
    #pragma GCC diagnostic ignored "-Wunused-value"
    #pragma GCC diagnostic ignored "-Wunused-label"
    #pragma GCC diagnostic ignored "-Wchar-subscripts"
    #pragma GCC diagnostic ignored "-Wmaybe-uninitialized"

    // ( ... )

    #pragma GCC diagnostic pop
```

the reasons of those warnings still need to be fixed, but this will unblock any situation where the IDE/compiler treats warnings as errors.

siara-cc, 2020-02-15:
Great findings @tobozo. I will add them.

tobozo, 2020-02-15:
I forgot specify that `#pragma GCC diagnostic pop` should be at the end of the file. 

This wasn't obvious to me when I tried it first anyway :-)

siara-cc, 2020-02-15:
Yes I understood that the first time.. Thanks.