siara-cc/esp32_arduino_sqlite3_lib #19: Mongoose os and nliviu`s sdlib crashed

namoaton, 2020-01-21:
Hello
I am trying to use your library with  mongoose os and their sdcard library and when I open db file from card I get 
```Guru Meditation Error: Core  0 panic'ed ```

siara-cc, 2020-01-22:
I will try it out and inform you. Thanks.

siara-cc, 2020-01-22:
This is the first time I am trying Mongoose OS.  Is it possible to share me the sketch or whatever is needed to compile this?

Also the Sqlite library is a memory hogger.  The problem you are facing is most likely because of that, given there is an additional OS layer to accommodate.  You may be better off using my other library https://github.com/siara-cc/sqlite_micro_logger_arduino instead for this.

namoaton, 2020-01-22:
I think that micro_logger project is not suitable for my project. Because I need to make some complex queries. So it would be great to be able to use sqlite3 library.
I use https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/open-source-hardware board. And I think that you can use any other board.

[sd_sqlite.zip](https://github.com/siara-cc/esp32_arduino_sqlite3_lib/files/4095674/sd_sqlite.zip)
unpack. enter to en_project directory.
```
mos build --clean --local --verbose --platform esp32
mos flash
```

namoaton, 2020-01-22:
I use ascii symbols for filename. Could be this cause of problem?

namoaton, 2020-01-22:
When use ESP IDF based frameworks don`t forget add
```
sqlite3_initialize();
```

closed 2020-01-22 by namoaton

thbl, 2020-08-15:
@namoaton could you share a small example / tutorial on how you got this to work with mongoose os Thanks :)