siara-cc/esp32_arduino_sqlite3_lib #23: disk I/O error on transaction with two tables on release 2.3

deividi21, 2020-03-30:
Hello @siara-cc ,

I've updated my library with the latest of your releases and found out that the same error that was occurring last month, solved on issue #21 , is now happening again.

By comparing the two versions, I've noticed that if I comment this line of code
`#define SQLITE_DEFAULT_CACHE_SIZE           -1`
on config_ext.h, it works, at least for my case, at least for now.

Please, take a look whenever you can. Thank you.

phamminhkhac, 2020-04-03:
I don't known where should I put my question, so maybe here is the good one.
I just want to ask "what happen if I insert new row into table but at the moment sqlite3 writen to bad cell on the Flash chip, does sqlite3 return "IO disk error" or retry to write again on another cell?

winkelict, 2023-12-08:
i think the solution in this issue might fix this problem https://github.com/siara-cc/esp32_arduino_sqlite3_lib/issues/81