siara-cc/esp32_arduino_sqlite3_lib #25: [Question] How to properly escape quoted strings to avoid SQL errors and/or injections ?

tobozo, 2020-04-20:
I have a few SQL insertion queries that will fail, whether I use single or double quotes along with unescaped contents.

I've been using Arduino's String.replace() for a while now but I feel this is as slow as it is unappropriate.

Back to my PHP+MySQL experience, I remember dealing with functions such as mysql::escape or mysqli::prepare that made such operation easier to deal with.

Is there such a thing as sqlite_escape_query() or sqlite_escape_value() ?

siara-cc, 2020-04-20:
Hi @tobozo,
You can't compare desktop/server PHP performance with Arduino, even if you are using ESP32, because of the difference in amount of RAM and CPU speed.

This escaping will be slower in any system, because of reallocation required - a longer string needs to be allocated to accommodate escaped characters, discarding the older string.  Only it is not apparent in PCs.

It can be made faster by pre-allocating target length of SQL using [reserve()](https://www.arduino.cc/reference/en/language/variables/data-types/string/functions/reserve).

Besides you may be interested in this [comparison](https://stackoverflow.com/a/16112845/5072621)

tobozo, 2020-04-20:
So does it mean String.replace() is the correct way to go ?

Sorry about my sloppy PHP metaphor :-))

siara-cc, 2020-04-20:
Yes string.replace() should be ok if used after reserve()

tobozo, 2020-04-20:
ok thanks, closing this as solved for me

closed 2020-04-20 by tobozo