siara-cc/esp32_arduino_sqlite3_lib #27: slow insertion - how to improve speed if need to insert large amount of rows ?

our30K, 2020-04-23:
Hi There

thanks for creating this neat sqlite for ESP.
quick question, I realise the insertion is kind of slow, 
how do I ensure fast insertion yet does not run out memory ?

I am using this to log sensor data, so it can be easily retrieve and manage later.
the sensor is pumping out 4-5 lines(100 bytes a line) of data every second. 
and since I have other things to do when dealing with this output, I will prefer the logging to be as fast as possible. 

Justin

tobozo, 2020-04-23:
Maybe use a 1K buffer to store the output, then only write when the buffer is nearly full, or once every 30 seconds if the buffer isn't full ?

SPIFFS blocks are 4K and minimal efficient SD write is 512 bytes, getting closer to those values when writing may improve the performances,

siara-cc, 2020-04-23:
@our30K 
Please try my other library https://github.com/siara-cc/sqlite_micro_logger_arduino, which exactly fits your need.  It was built for the purpose of logging sensor data faster than the official library.

@tobozo 
What you have suggested is just what the Sqlite Micro Logger does.  You could choose the block size depending on the optimal size for the FS.

Also you could use any IO Library - SPIFFS, SD, SdFat etc.  I am planning to do the same for the library for ESP8266, instead of having to depend on SdFat (well, when I get time :-)).

our30K, 2020-04-24:
thanks, I will give it a try in the next few days. 
meanwhile, I assume this also solve the problem of slow down when table gets big issue ?

rjjrbatarao, 2021-07-08:
@our30K  I used sqlite transactions to isert faster here is some part of my code, `  sqlite3_exec(db, "begin;", 0, 0, &zErrMsg); //Open transaction
  JsonArray array = json.as<JsonArray>();
  for (JsonVariant v : array) {
    sprintf_P(sql_statement, PSTR("UPDATE rates SET price=%s,time=%s,data=%s,charge=%s,profile='%s' WHERE price=%s;"), v["price"].as<char*>(), v["time"].as<char*>(), v["data"].as<char*>(), v["charge"].as<char*>(), v["profile"].as<char*>(), v["price"].as<char*>());
    rc = sqlite3_exec(db, sql_statement, web_callback, 0, &zErrMsg);
    memset(sql_statement, 0, 255);
    if ( rc != SQLITE_OK)
    {
      is_exec_success = false;
      Serial.printf("SQL ERROR: %s\n", zErrMsg);
      sqlite3_free(zErrMsg);
      web_request = WEB_FAILED;
    } else {
      web_request = WEB_IDLE;
    }
  }
  if (!is_exec_success) {
    sqlite3_exec(db, "rollback;", 0, 0, 0);
  } else {
    sqlite3_exec(db, "commit;", 0, 0, 0);
  }
  sqlite3_close(db);`