siara-cc/esp32_arduino_sqlite3_lib #32: Can i host a webpage from spiffs if i use sqlite3_lib?

zekageri, 2020-09-01:
I'am wondering if it is possible to host a webpage from flash if i use this sql lib.
For example if i use async http requests like this:

```
server.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
    AsyncWebServerResponse* response = request->beginResponse(LITTLEFS, "/Home_index.html", "text/html");
    response->addHeader("Content-Encoding", "gzip");
    request->send(response);
});
```

Will it work if i init the sqlite3?

siara-cc, 2020-12-19:
This library works with requests from WiFi. But I don't understand your question. You want to serve static pages stored in SQLite database?

zekageri, 2020-12-20:
Yes kindof. Or just make part of the flash to sqlite.

closed 2022-08-23 by zekageri