siara-cc/esp32_arduino_sqlite3_lib #34: How long does it take to insert the data?

Rita-GE, 2020-09-08:
Hi,
I wonder how long does ESP32 take to insert the data?

siara-cc, 2020-09-11:
I think you have index in your database and whenever insertion happens in table index also has to be updated. Thats probably why it is slow.
To solve the problem you could try tables using WITHOUT ROWID keyword and use the columns that need to be indexed as primary keys.

Rita-GE, 2020-09-16:
> I think you have index in your database and whenever insertion happens in table index also has to be updated. Thats probably why it is slow.
> To solve the problem you could try tables using WITHOUT ROWID keyword and use the columns that need to be indexed as primary keys.

But the thing is I have never used index in my table.

GithubPatrick91, 2020-09-19:
I have the same issue.
Even when i try WITHOUT ROWID, its not faster.
Sometimes the INSERT Transaction takes around 27seconds for just 1000 integer.

siara-cc, 2020-09-19:
@Rita-GE @GithubPatrick91 
I will try on my side.  Meanwhile, if you are trying to store sensor data, this library:
https://github.com/siara-cc/sqlite_micro_logger_arduino
is more suitable.  It can both read and write SQLite dbs, but it does not support SELECT, INSERT etc. You will have to use the given API. Please see the examples.

GithubPatrick91, 2020-09-20:
I already tried this library - and like your lightweigth and fast µlogger lib. 

But unfortunately the speed of the "finalize()" command decrease also continuously. 
When the database has around 1 million rows (3 columns) the finalize takes around ~13seconds.

(info: i used the SELECT command of your sqlite3 library to display the table)

siara-cc, 2020-12-19:
Have you tried using Prepared Statement to insert the records?

ram-prasanth-9394, 2021-01-07:
Hi @siara-cc 
It was very useful to use your code 

But i need your help 
I was inserting and updating large amount of data using sd card in sqlie3 db 
I was facing some errors in the SELECT query and will post the code below.

**Code**
`#include <SD.h>

#include <stdio.h>

#include <stdlib.h>

#include <sqlite3.h>

#include <FS.h>

#include <WiFi.h>

#include <HTTPClient.h>

#include <ArduinoJson.h>

const char * host = "API KEY";

const char * wifiName = "******";
const char * wifiPass = "******";
bool Flag = false;
int rc;
sqlite3 * db;

const char * data = "Callback function called";
static int callback(void * data, int argc, char ** argv, char ** azColName) {
  int i;
  Serial.printf("%s: ", (const char * ) data);
  for (i = 0; i < argc; i++) {
    Serial.printf("%s = %s\n", azColName[i], argv[i] ? argv[i] : "NULL");
  }
  Serial.printf("\n");
  return 1;
}
void listDir(fs::FS & fs,
  const char * dirname, uint8_t levels) {
  Serial.printf("Listing directory: %s\n", dirname);

  File root = fs.open(dirname);
  if (!root) {
    Serial.println("Failed to open directory");
    return;
  }
  if (!root.isDirectory()) {
    Serial.println("Not a directory");
    return;
  }

  File file = root.openNextFile();
  while (file) {
    if (file.isDirectory()) {
      Serial.print("  DIR : ");
      Serial.println(file.name());
      if (levels) {
        listDir(fs, file.name(), levels - 1);
      }
    } else {
      Serial.print("  FILE: ");
      Serial.print(file.name());
      Serial.print("  SIZE: ");
      Serial.println(file.size());
    }
[... 219 lines cut ...]
**HERE i will print API data **
Opened database successfully
CREATE TABLE IF NOT EXISTS NFCCardDetails (Physicalid String PRIMARY KEY ,WalletAmount Float)
Operation done successfully
SELECT * FROM NFCCardDetails where Physicalid='62:8a:2f:2e'
Callback function called: Physicalid = 62:8a:2f:2e
WalletAmount = 205.0

SQL error: query aborted
4
Updating Value
UPDATE NFCCardDetails SET WalletAmount = '205' WHERE Physicalid = '62:8a:2f:2e'
Operation done successfully
SELECT * FROM NFCCardDetails where Physicalid='62:8a:2f:2e'
Callback function called: Physicalid = 62:8a:2f:2e
WalletAmount = 205.0

SQL error: query aborted
4
Updating Value
UPDATE NFCCardDetails SET WalletAmount = '205' WHERE Physicalid = '62:8a:2f:2e'
Operation done successfully
SELECT * FROM NFCCardDetails where Physicalid='04:6b:43:3a:ec:64:80'
Callback function called: Physicalid = 04:6b:43:3a:ec:64:80
WalletAmount = 88.0