siara-cc/esp32_arduino_sqlite3_lib #37: UNION or JOINS in views

tomthemod, 2020-11-20:
This library is very very helpfull for my projects.
But i need to create views with UNION, or LEFT JOIN. 
When i create a view like "CREATE VIEW view1 AS SELECT id, data AS tbl1 UNION SELECT id, data AS tbl2;" directly from the sqlite3_console example with a database with 2 tables, i've this error :
SQL error: 1 parser stack overflow

I've the same error if i create the view in the database from my computer, succesfully test it, and try to use the database from the ESP (esp32)

Is there an option to activate, to do this ?

Thnaks.

siara-cc, 2020-11-21:
You could try changing the stack settings in the config file of this library. But I don't this they would work seamlessly given the limited memory available to operate.

tomthemod, 2020-11-22:
thanks for your support, you've really made a good job.
I've finally reloved my issue with increasing YYSTACKDEPTH in config_ext.h, from 20 to 50.
Thanks.

MantaRayDeeJay, 2023-05-19:
I met the same issue.
I found the following link [YYSTACKDEPTH — Maximum parser stack depth](https://www.oreilly.com/library/view/using-sqlite/9781449394592/re09.html)
It says that a value of 25 for the YYSTACKDEPTH param should be enough on common usages.
I have tested with a value of 25 and it seems being enough to work.