siara-cc/esp32_arduino_sqlite3_lib #38: Callback not called after searching non existent record

albertlt, 2020-12-06:
I have a database containing 1 table with 1 primary key and 2 fields. I inserted 1016 records in total. Everything working great and fast except when I search something that is not inside the 1016 records. For example: SELECT * FROM mytable WHERE myPK ='NOT-IN-REC';

Yes, the primary key is VARCHAR(20). The db_exec function would execute successfully and I can see Serial out: "Operation done successfully" followed with "Time taken:42632".

However, the callback function was never called. If I search something that is in the database, then the callback function will be called. Any solution to this issue?

albertlt, 2020-12-06:
Further update. I tried issuing: INSERT OR REPLACE on one of the record and now the callback is working. However, the search result is as follow:

18:53:46.230 -> Opened database successfully
18:53:46.230 -> SELECT * FROM accesslist WHERE key='7A2AB925'
18:53:46.230 -> E (122640) vfs_fat: open: no free file descriptors
18:53:46.230 -> E (122643) vfs_fat: open: no free file descriptors
18:53:46.230 -> SQL error: unable to open database file
18:53:46.230 -> Time taken:11157

albertlt, 2020-12-09:
Found out that callback was not called on empty query (SELECT something that does not exist). How can I still call the callback to notify the empty query result?

siara-cc, 2020-12-19:
I don't think callback can be called for empty result as per SQLite API.  You could maintain a variable counter to check how many records or use SELECT COUNT(*) query before you execute your actual query.  Also there are other examples in the repo where callback if not needed (using sqlite3_step() function).

albertlt, 2020-12-20:
Yeah I figured it out already. Thank you for your great work. Really appreciate it. Cheers.