siara-cc/esp32_arduino_sqlite3_lib #41: Temporary sort file, Watch dog

vvvait, 2021-01-26:
Hi.
Thanks for adapting sqlite3 for esp32.
I researched the work of the library for several months.
On some requests, I began to receive the "disk I / O error" error. I got into the source and found that there is no way to create temporary files.
in esp32.cpp:  ESP32Open
```
const char *zName,              /* File to open, or 0 for a temp file */
...
if( zName==0 ){
  return SQLITE_IOERR;
}

```
i replaced `return SQLITE_IOERR; ` to `zName = "/sd/temp";` 
and selection of 1000 records with sorting began to work.

It is also worth adding feeding the WatchDog on requests longer than 3 seconds if execute worked in main loop task, otherwise it will fire, may by in ESP32Read and ESP32Write, or add some callbacks.

max performance i have with
#define SQLITE_DEFAULT_PAGE_SIZE           512 //4096
#define SQLITE_DEFAULT_CACHE_SIZE           -8 // -1

siara-cc, 2021-01-30:
Hi, Thank you very much for communicating your findings.  I will incorporate these into the repo.

vvvait, 2021-01-30:
Hi, you still need to delete the temporary file if it already exists, otherwise an error is displayed periodically that the database is damaged
```
if( zName==0 ){
    zName = "/sd/temp";
    remove(zName); 
}
```

winkelict, 2023-12-08:
this might be a full implementation of temporary file support:
https://github.com/siara-cc/esp32_arduino_sqlite3_lib/issues/81

siara-cc, 2023-12-31:
@vvvait I have incorporated this with today's commit using code from @winkelict and @savejeff 
Thanks to all of you for the support!