siara-cc/esp32_arduino_sqlite3_lib #42: SQLite 2.3.0 - DELETE with Subquery returns [SQL error: parser stack overflow]

genotix, 2021-01-27:
Hi,

I've been trying to execute a DELETE to clean up a single table which fails with error: **SQL error: parser stack overflow**

`
sql = F("DELETE FROM raw_log WHERE rowid IN (SELECT rowid FROM raw_log limit 20 ASC);"); // Remove oldest first   
   int rc = db_exec(db1, sql.c_str() );
   if (rc != SQLITE_OK) {
       sqlite3_close(db1);
       return;
   }      
   vacuum();
`

The table structure:
`CREATE TABLE IF NOT EXISTS raw_log (reg DATETIME, transmitted BOOL, content BLOB);`

Any hints on what I'm doing wrong would be appreciated.
(I could the the select separately and run a loop to delete per record and would expect this to work but I rather have clean code).

Dhruvan1, 2023-03-23:
> Hi,
> 
> I've been trying to execute a DELETE to clean up a single table which fails with error: **SQL error: parser stack overflow** My device is an ESP32-WROVER-B (TTGO T-CALL v1.4 to be exact) and I'm using the SPIFFS library without any SD card.
> 
> **The snippet of code responsible for the delete action:** `sql = F("DELETE FROM raw_log WHERE rowid IN (SELECT rowid FROM raw_log limit 20 ASC);"); // Remove oldest first int rc = db_exec(db1, sql.c_str() ); if (rc != SQLITE_OK) { sqlite3_close(db1); return; } vacuum();`
> 
> **The table structure:** `CREATE TABLE IF NOT EXISTS raw_log (reg DATETIME, transmitted BOOL, content BLOB);`
> 
> Any hints on what I'm doing wrong would be appreciated. (I know could execute the the select separately and run a loop to delete per record and would expect this to work but I rather have clean code if possible; SQLite seems to be -very- complete given the lightweight system it is running on).
 
 
 facing same issue

siara-cc, 2023-03-23:
Hi, please see this.  I think vacuum is causing this issue. I am not sure if there is a simple solution.  Best solution I can think of is to maintain periodical logs and delete the old files.
https://github.com/siara-cc/esp32_arduino_sqlite3_lib/issues/50#issuecomment-916045739

siara-cc, 2023-03-23:
Also https://github.com/siara-cc/esp32_arduino_sqlite3_lib/issues/43#issuecomment-1131262737