siara-cc/esp32_arduino_sqlite3_lib #46: Need help with insert statement

happytm, 2021-06-03:
Arun,

I am testing your excellent library and works fine with static data insert and select queries as tested with following sketch:

https://github.com/happytm/BatteryNode/blob/master/test/SQLite3.ino

Now I tried to use same code in larger project where data has to be inserted in to SQLite3 database dynamically from remote devices. My project receives data fine from remote devices but it fails to insert data in to database. The project sketch is at link below:

https://github.com/happytm/BatteryNode/blob/master/test/SQLite3_Proberequest.ino

Can you please look at both code with your expert eye and see where I am making mistake?

Thanks.

happytm, 2021-06-03:
Never Mind. Upon further review I found my mistake and now it is working fine.

Sorry for the trouble.

Thanks.

closed 2021-06-03 by happytm

siara-cc, 2021-06-05:
Hi @happytm  Sorry I can work on these only on weekends so could not reply to you earlier.  Glad that you got that figured out.