siara-cc/esp32_arduino_sqlite3_lib #47: ESP32 panics with certain code.

happytm, 2021-06-04:
Hi Arun ,

I was able to implement your library with Asyncwebserver, Websockets and TinyMQTT and it works as expected. I was able to insert Fake data created by function called processData() and also I can query data from database using same function by requesting data from websocket client. 

When I call processData() function from loop it works fine but when I call same function from another place (line 542) it panics and decoded message looks like following:

```
PC: 0x4008c01b
EXCVADDR: 0x00000000

Decoding stack results
0x4018a80f: esp_ipc_call_and_wait at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/ipc.c line 116
0x4018a8e6: esp_ipc_call at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/ipc.c line 123
0x400871ce: spi_flash_disable_interrupts_caches_and_other_cpu at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c line 122
0x400876d1: spi_flash_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c line 165
0x40191a17: esp_partition_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/partition.c line 242
0x4011d070: littlefs_api_read at C:\Users\WinCycle\Documents\Arduino\libraries\LittleFS_esp32\src\littlefs_api.c line 22
0x40119cf5: lfs_bd_read at C:\Users\WinCycle\Documents\Arduino\libraries\LittleFS_esp32\src\lfs.c line 104
0x4011a14f: lfs_dir_fetchmatch at C:\Users\WinCycle\Documents\Arduino\libraries\LittleFS_esp32\src\lfs.c line 861
0x4011a705: lfs_dir_find at C:\Users\WinCycle\Documents\Arduino\libraries\LittleFS_esp32\src\lfs.c line 1245
0x4011cbb1: lfs_stat at C:\Users\WinCycle\Documents\Arduino\libraries\LittleFS_esp32\src\lfs.c line 3189
0x40118f15: vfs_littlefs_fstat at C:\Users\WinCycle\Documents\Arduino\libraries\LittleFS_esp32\src\esp_littlefs.c line 1165
0x40187f71: esp_vfs_fstat at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs.c line 479
0x40148be9: fstat at ../../../.././newlib/libc/syscalls/sysfstat.c line 12
0x40116f9f: ESP32FileSize(sqlite3_file*, sqlite_int64*) at C:\Users\WinCycle\Documents\Arduino\libraries\Sqlite3Esp32\src\esp32.cpp line 305
0x4019958f: sqlite3OsFileSize at C:\Users\WinCycle\Documents\Arduino\libraries\Sqlite3Esp32\src\sqlite3.c line 22896
0x400d5add: pagerPagecount at C:\Users\WinCycle\Documents\Arduino\libraries\Sqlite3Esp32\src\sqlite3.c line 54718
0x400f1ca1: sqlite3PagerSharedLock at C:\Users\WinCycle\Documents\Arduino\libraries\Sqlite3Esp32\src\sqlite3.c line 56848
0x400f4723: sqlite3BtreeBeginTrans at C:\Users\WinCycle\Documents\Arduino\libraries\Sqlite3Esp32\src\sqlite3.c line 67181
0x40108145: sqlite3InitOne at C:\Users\WinCycle\Documents\Arduino\libraries\Sqlite3Esp32\src\sqlite3.c line 127174
0x4010e225: sqlite3Init at C:\Users\WinCycle\Documents\Arduino\libraries\Sqlite3Esp32\src\sqlite3.c line 127359
0x4010e282: sqlite3ReadSchema at C:\Users\WinCycle\Documents\Arduino\libraries\Sqlite3Esp32\src\sqlite3.c line 127385
0x4010f6f0: sqlite3LocateTable at C:\Users\WinCycle\Documents\Arduino\libraries\Sqlite3Esp32\src\sqlite3.c line 109489
0x4010f87d: sqlite3LocateTableItem at C:\Users\WinCycle\Documents\Arduino\libraries\Sqlite3Esp32\src\sqlite3.c line 109550
0x4011021a: sqlite3SrcListLookup at C:\Users\WinCycle\Documents\Arduino\libraries\Sqlite3Esp32\src\sqlite3.c line 114722
0x40112c79: sqlite3Insert at C:\Users\WinCycle\Documents\Arduino\libraries\Sqlite3Esp32\src\sqlite3.c line 119829
0x40105a75: sqlite3RunParser at C:\Users\WinCycle\Documents\Arduino\libraries\Sqlite3Esp32\src\sqlite3.c line 156261
0x40107da5: sqlite3Prepare at C:\Users\WinCycle\Documents\Arduino\libraries\Sqlite3Esp32\src\sqlite3.c line 127586
0x40107f32: sqlite3LockAndPrepare at C:\Users\WinCycle\Documents\Arduino\libraries\Sqlite3Esp32\src\sqlite3.c line 127658
0x40107f75: sqlite3_prepare_v2 at C:\Users\WinCycle\Documents\Arduino\libraries\Sqlite3Esp32\src\sqlite3.c line 127742
0x4010db3f: sqlite3_exec at C:\Users\WinCycle\Documents\Arduino\libraries\Sqlite3Esp32\src\sqlite3.c line 122065
0x400d20c4: db_exec(sqlite3*, char const*) at C:\Users\WinCycle\Documents\Arduino\sqlite3.ino/sqlite3.ino.ino line 167
0x400d2637: processData() at C:\Users\WinCycle\Documents\Arduino\sqlite3.ino/sqlite3.ino.ino line 371
0x400d2dc0: probeRequest(system_event_id_t, system_event_info_t) at C:\Users\WinCycle\Documents\Arduino\sqlite3.ino/sqlite3.ino.ino line 542
0x400d19c9: std::_Function_handler ::_M_invoke(std::_Any_data const&, system_event_id_t&&, system_event_info_t&&) at c:\users\wincycle\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0\xtensa-esp32-elf\include\c++\5.2.0/functional line 1871
0x400d461d: WiFiGenericClass::_eventCallback(void*, system_event_t*, wifi_prov_event_t*) at c:\users\wincycle\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0\xtensa-esp32-elf\include\c++\5.2.0/functional line 2271
0x400d46f8: _network_event_task(void*) at C:\Users\WinCycle\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi\src\WiFiGeneric.cpp line 73
0x4008993e: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```
My sketch is at: https://github.com/happytm/BatteryNode/blob/master/test/SQLite3_Proberequest.ino

Can you please help me resolve this issue?

siara-cc, 2021-06-05:
- Usually event callbacks are time bound and expect the handler to return within a reasonable interval. If not returned before the next event, it could lead to cascading calls and fill up the stack and eventually blow over.  So including processData() could be leading to this.
- Please check the allocation of 256 to str and 70 to s.  Any possibility of under allocation?

happytm, 2021-06-05:
I doubled the allocation for both but having same problem.

Thanks.

happytm, 2021-06-06:
I even tried almost empty function like below:

void probeRequest(WiFiEvent_t event, WiFiEventInfo_t info) {  
           processData();
}

And ESP32 panic as soon as it reaches processData();

It is surprising that processData work fine when it is called from loop but behave differently when called from other places.

My error message in arduino serial console is as follows:
```
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (network_event) 
Core 1 register dump:
PC      : 0x4008c018  PS      : 0x00060236  A0      : 0x40086e14  A1      : 0x3ffb85d0  
A2      : 0x00000001  A3      : 0x3ffc8728  A4      : 0x3ffc8728  A5      : 0x00000001  
A6      : 0x00060220  A7      : 0x00000000  A8      : 0x80085428  A9      : 0x3ffb8670  
A10     : 0x3ff000e0  A11     : 0x00000001  A12     : 0x3ffbfbe8  A13     : 0x00000001  
A14     : 0x00060223  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x4008c018:0x3ffb85d0 0x40086e11:0x3ffb86b0 0x4018a69b:0x3ffb86f0 0x4018a772:0x3ffb8710 0x400871ce:0x3ffb8730 0x400876d1:0x3ffb8750 0x401918a3:0x3ffb87c0 0x4011cefc:0x3ffb87f0 0x40119b81:0x3ffb8820 0x40119fdb:0x3ffb8850 0x4011a591:0x3ffb88f0 0x4011ca3d:0x3ffb8950 0x40118da1:0x3ffb89a0 0x40187dfd:0x3ffb8ad0 0x4000bcdd:0x3ffb8af0 0x40148a75:0x3ffb8b10 0x40116e63:0x3ffb8b30 0x4019941b:0x3ffb8b90 0x400d59c9:0x3ffb8bb0 0x400f1b8d:0x3ffb8be0 0x400f460f:0x3ffb8c30 0x40108031:0x3ffb8c80 0x4010e111:0x3ffb8d10 0x4010e16e:0x3ffb8d30 0x4010f5b4:0x3ffb8d50 0x4010f741:0x3ffb8d90 0x401100de:0x3ffb8db0 0x40112b3d:0x3ffb8dd0 0x40105961:0x3ffb8f30 0x40107c91:0x3ffb9110 0x40107e1e:0x3ffb9260 0x40107e61:0x3ffb92a0 0x4010da2b:0x3ffb92d0 0x400d2108:0x3ffb9320 0x400d264b:0x3ffb9340 0x400d2d60:0x3ffb93b0 0x400d19c5:0x3ffb93f0 0x400d4509:0x3ffb9440 0x400d45e4:0x3ffb9530 0x4008993e:0x3ffb9560
```
"Debug exception reason: Stack canary watchpoint triggered (network_event) " this message in particular definitely points to proberequest function.

Thanks.

siara-cc, 2021-06-06:
Hi, my opinion is that this is happening because processData has a lot of calls to Sqlite API functions and each call takes time in the event handler is holding things up.

Is it possible to set a flag in the event handler and return immediately and call processData from  loop() function based on the flag and reset the flag after that?  You may have to call the MQTT loop() and client poll() function at each step of processData() so as not to loose packets from MQTT or WiFi.

happytm, 2021-06-06:
That's what I was thinking about doing. Somehow use loop to trigger processData() but I am not sure how to achieve that.

Thanks.

siara-cc, 2021-06-06:
Just add a flag such as `int triggerProcessData = 0` on top.  Set the value to 1 in the event handler and add `if (triggerProcessData) { processData(); triggerProcessData  = 0; }` in loop function.

happytm, 2021-06-07:
Thank you very much for your help.

You are absolutely on spot diagnosing the problem and providing accurate solution for it.

ESP32 is happy now and does not panic anymore with insert statement.

My new issue is Websocket server code in loop is blocking one which does not allow any code until  websocket client is connected. I  am going to raise issue at their repository to find out way around it. My loop looks like below:

```
void loop() {
if (triggerProcessData == 1) { 
    processData(); 
    triggerProcessData = 0; 
    } 
    
#if MQTT
  broker.loop();  // Don't forget to add loop for every broker and clients
  myClient.loop();
#endif
  
#if WEBSOCKETS
  //auto client = WSserver.accept();
  //client.onMessage(handle_message);
  //processData();
  static auto next=millis();               // The next line is an efficient delay() replacement
  if (millis() > next){next += 10000;}       //let's give time to SQLITE3 to prepare data requested from client.
  
  //while (client.available()) {
  //client.poll();
  //}
#endif  
} // End of loop
```
This line "auto client = WSserver.accept();" in code above is blocking one so it is stuck there.

Thanks again.

happytm, 2021-06-08:
I finally got the issue with websockets blocking the code in loop resolved with library's creator. 

Now everything works as expected. So this issue is resolved.

Thank you for your help and patience.