siara-cc/esp32_arduino_sqlite3_lib #48: How to prepare query response String to be written to .jason file

happytm, 2021-06-07:
I wanted to write query response String to a .json file for further processing of data at web interface.

I tried to modify callback function like below:

```
const char* data = "Data requested";
 static int callback(void *data, int argc, char **argv, char **azColName) {
 int i;
  for (i = 0; i<argc; i++){
      Serial.printf("%s,", argv[i] ? argv[i] : "NULL");   
      String resp = argv[i];
      Serial.print(resp);
      }
       Serial.printf("\n");
       return 0;  
}
```
Which produces result like below:

```
1623049821,162304982124785,247851535,153515,1526,2628,2830,3055,5565,65230,23083,83
1623049826,162304982624785,247851535,153515,1526,2628,2830,3055,5565,65230,23083,83
1623049834,162304983424785,247851535,153515,1526,2628,2830,3055,5565,65230,23083,83
```
But I wanted to write to a json file like below:
```
[
11,"Timestamp","Date","time","Hour","Location","Voltage","RSSI","Temperature","Humidity","Pressure","Light",
1623049821,24785,1535,15,26,28,30,55,65,230,83,
1623049826,24785,1535,15,26,28,30,55,65,230,83,
1623049834,24785,1535,15,26,28,30,55,65,230,83
]
```
Can you please show me how to create a String to produce above result?

Thanks.

happytm, 2021-06-08:
If possible, result in following format is even better. I want to feed this directly into uPlot graphing library.

```
[
[1623049821,1623049826,1623049834],
[24785,24785,24785],
[1535,1535,1535],
[15,15,15],
[26,26,26],
[28,28,28],
[30,30,30],
[55,55,55],
[65,65,65],
[230,230,230],
[83,83,83]
]

Thank you.

siara-cc, 2021-06-12:
Please see the example webquery in this repo. Using the prepare statement and step sqlite api functions, you could form such strings as above:

```
      while (sqlite3_step(res) == SQLITE_ROW) {
```

happytm, 2021-06-13:
Hi Arun,

Based on your guidance I was able to implement following code:

```
Serial.print("SQLite rows: "); Serial.println(SQLITE_ROW);
   
while (sqlite3_step(res) == SQLITE_ROW) {
          
          resp += "[";
          resp += sqlite3_column_int(res, 0);
          resp += ",";
          resp += sqlite3_column_int(res, 1);
          resp += ",";
          resp += sqlite3_column_int(res, 2);
          resp += ",";
          resp += sqlite3_column_int(res, 3);
          resp += ",";
          resp += sqlite3_column_int(res, 4);
          resp += ",";
          resp += sqlite3_column_int(res, 5);
          resp += ",";
          resp += sqlite3_column_double(res, 6);
          resp += ",";
          resp += sqlite3_column_int(res, 7);
          resp += ",";
          resp += sqlite3_column_int(res, 8);
          resp += ",";
          resp += sqlite3_column_int(res, 9);
          resp += ",";
          resp += sqlite3_column_int(res, 10);
          resp += "],";
          Serial.println(resp);
          rec_count++;
      }
      resp = "Number of records: ";
      resp += rec_count;
      Serial.println(resp);
      sqlite3_finalize(res);
      sqlite3_close(db1);
   ```
Which produces result like below:

```
Number of records: 2[1623562895,24785,1535,15,6,-39,2.92,46,71,214,82],
Number of records: 2[1623562895,24785,1535,15,6,-39,2.92,46,71,214,82],[1623562910,24785,1535,15,26,-25,2.94,54,57,244,82],
Number of records: 2[1623562895,24785,1535,15,6,-39,2.92,46,71,214,82],[1623562910,24785,1535,15,26,-25,2.94,54,57,244,82],[1623563044,24785,1535,15,6,-38,2.92,53,66,252,41],
Number of records: 2[1623562895,24785,1535,15,6,-39,2.92,46,71,214,82],[1623562910,24785,1535,15,26,-25,2.94,54,57,244,82],[1623563044,24785,1535,15,6,-38,2.92,53,66,252,41],[1623563060,24785,1535,15,6,-38,2.92,53,66,252,41],
Number of records: 2[1623562895,24785,1535,15,6,-39,2.92,46,71,214,82],[1623562910,24785,1535,15,26,-25,2.94,54,57,244,82],[1623563044,24785,1535,15,6,-38,2.92,53,66,252,41],[1623563060,24785,1535,15,6,-38,2.92,53,66,252,41],[1623563194,24785,1535,15,6,-37,2.92,52,55,239,62],
Number of records: 2[1623562895,24785,1535,15,6,-39,2.92,46,71,214,82],[1623562910,24785,1535,15,26,-25,2.94,54,57,244,82],[1623563044,24785,1535,15,6,-38,2.92,53,66,252,41],[1623563060,24785,1535,15,6,-38,2.92,53,66,252,41],[1623563194,24785,1535,15,6,-37,2.92,52,55,239,62],[1623563210,24785,1535,15,6,-37,2.92,52,55,239,62],
Number of records: 8
```
Is it possible to get values of columns in same rows and thereby have 11 rows each containing values of same columns ?

If possible can you please give me example? Sorry I do not understand loops very well.

Thanks

siara-cc, 2021-06-13:
I am unable to spend time on what you are trying to achieve, but it looks like just by fiddling with the code in different ways you will be able to figure it out.

happytm, 2021-06-13:
Ok I will try to figure it out.

Thanks.