siara-cc/esp32_arduino_sqlite3_lib #50: SQLite Database Disk Image is Malformed

reshma26994, 2021-09-09:
i am using https://github.com/siara-cc/esp32_arduino_sqlite3_lib for store my data in SPIFFS of ESP32 but after some insertion of data i got error of SQLite Database Disk Image is Malformed

mr-wiggle, 2021-09-09:
This is an issue with the SPIFFS file system.   I started out using this library on SPIFFS and was getting Disk Image Malformed errors all over the place.   I switched to LITTLEFS, which is a drop in replacement, and the errors just went away.

https://github.com/lorol/LITTLEFS

PaulDK89, 2022-08-08:
If it helps, my db kept getting Malformed error when deleting data, and to fix i had to go down back to the esp 1.0.6 board lib. The 2.4 lib seemed to have caused it on SD card management side.