siara-cc/esp32_arduino_sqlite3_lib #55: VACUUM result disk I/O error

manouhihalid, 2022-02-03:
Hi,

I try to use "VACUUM" but i get this error disk I/O error.

Other sqlite command work well.

I use ESP32 and SPIFFS.

i use the command 

sqlcommand="VACUUM";
"db_exec(db1, sqlcommand);"

thanks for your support

uknow19, 2022-03-02:
Did you manage to resolve this? I'm getting the same issue.

siara-cc, 2022-03-02:
Hi, not sure what is causing this, but please check this out: https://github.com/siara-cc/esp32_arduino_sqlite3_lib/issues/50#issuecomment-916045739

PaulDK89, 2022-08-08:
Hi,

I am experiencing the same issue but through the SPI bound SD card. All other functions are fine, only VACUUM causing this issue.

savejeff, 2023-05-10:
I have looked into this and its solvable:

1.) a temporary file is opened. this is not implemented currently -> filename is 0 and the doc says you have to come up with a filename yourself. -> first thing that needs to be fixed in ESP32Open like this: 
```

if( zName==0 ) {
		//return SQLITE_IOERR;
		zName = "/spiffs/.tmp";
}

```

2.) next the temp file needs to be initialized with some data as when writing out the seek somehow fails. 
i have not solved this correctly but this helps somehow: 

```

		FILE* fp = fopen(zName, "w+");
		uint8_t buff[128] = { 0 };
		memset(buff, 0, sizeof(buff));
		for(int i = 0; i < 4096 * 2 / 128; i++)
			fwrite(buff, 1, sizeof(buff), fp);
		fclose(fp);
		
```
		

lastly, truncate is not implemented and thus the operation sometimes fails. ESP32 for an unknown reason refuses to implement truncate in there filesystem. you can work around this by (moving the file -> creating a new file with the name of the original file with the correct truncated size -> copy data from the original file into new file)

___

its not a step by step fix but helping hints for somebody how wants to fix this. my code has at this point deviated to far so that i can publish a pullrequest. 

also some other problems might also be fixed by implementing the temporary file and trucate like the issue with dropping tables or auto increment primary key

siara-cc, 2023-05-10:
@savejeff  Thanks for sharing this!!  I will check.

savejeff, 2023-05-11:
i think i have now fixed most of the problems that caused the IO Disk errors others reported. @siara-cc i can send you the my implementation. its based on arduinos FS classes and not on the basic fwrite, etc functions but it demonstrates the needed general behaviour

siara-cc, 2023-05-12:
Hi, thank you for the detailed explanation.  If a PR is not possible, I suggest attaching the zip file here so I can check and modify the library.

winkelict, 2023-12-08:
@savejeff ive posted an issue with a fix for esp32open
https://github.com/siara-cc/esp32_arduino_sqlite3_lib/issues/81