siara-cc/esp32_arduino_sqlite3_lib #65: sqlite3_prepare_v2 breaks client.connect()

sharkyenergy, 2022-11-21:
Hello, 

trying again to submit the issue today as it was not working 2 days ago.

first off, thank you for your amazing work.

`
rc = sqlite3_prepare_v2(db1, sql.c_str(), sql.length()+1, &res, &tail);
`

If I run above line, the client.connect() function used by AsyncTelegram2 api stops working.

```

AsyncTelegram2::AsyncTelegram2(Client &client, uint32_t bufferSize)
{
    m_botusername.reserve(32); // Telegram username is 5-32 chars lenght
    m_rxbuffer.reserve(bufferSize);
    this->telegramClient = &client;
    m_minUpdateTime = MIN_UPDATE_TIME;
}
.
.
.
if (!telegramClient->connect(TELEGRAM_HOST, TELEGRAM_PORT))
        {   
            Serial.println("\n\nUnable to connect to Telegram server");
        }
```

this code in particular is not running anymore after calling sqlite3_prepare_v2. it always returns unable to connect. If I remove your line of code, it works again.

I am on a ESP32 board.
Any idea of what might be the cause of it?
Thank you!

best regards

Igor

siara-cc, 2022-11-22:
Hi, I am unable to say why this is happening for you, but such issues are almost always related to memory leaks.  Please check for memory sufficiency / allocation / deallocation issues.  If possible please post entire code and schema and page size.  Please try 512 bytes page size, minimise index usage and [use tables without rowid (if index needed)](https://www.sqlite.org/withoutrowid.html) if not already doing so.

tanzerlana, 2022-11-22:
Thank you! will try those things today when I am back home. My code is rather long to post, but since this all happens in the first few lines of code in the setup, I can prepare a reproducible example in case i am not able to solve with the sugestions.

there is just the sd card init, then the sql init to get teh wifi login data, then the wifi init, and finally telegram. will post back with the result later today!

meanwhile thank you for your support!

sharkyenergy, 2022-11-23:
sorry for the late reply, my esp broke and i ordered a new one. will check it as soon as I get hte esp. i was wondering, how can i change the page size? i googeled but cant find anything...

siara-cc, 2022-11-25:
ok, to change page size please see: https://www.sqlite.org/pragma.html#pragma_page_size

However, this is better done with empty database if created from ESP32.  VACUUM is known to cause issues when there is data.

I just realised I had kept the default as 4096.  If above does not work please change it to 512 and compile before creating database:
https://github.com/siara-cc/esp32_arduino_sqlite3_lib/blob/0850945aa8f20504d4dc3ad3e9a7c36939e51fbc/src/config_ext.h#L28

siara-cc, 2022-11-25:
I changed the above default page size so you could download the latest master branch if you would like to try it out. The database would have to be created after uploading to ESP32