siara-cc/esp32_arduino_sqlite3_lib #68: Working with simultaneous actions

Zhu-jiatong, 2023-03-13:
I am using this library with a webserver for storing and accessing user data.
In most cases, there will be no issues.

However, when there is a write operation to the database while a read is being performed on it, what will happen?
What about 2 writes at the same time?

I saw that on the [sqlite documentation](https://sqlite.org/threadsafe.html), different threadsafe options are offered both during compile time and runtime. 
So, according to this document, I may use `Multi-thread` mode to allow a database to be used simultaneously by multiple connections.
Alternatively, I may also use `Serialized` mode to force all operations into sequential order.

I would like to know if the above informatin is valid, and that whether I can use the library safely in such cases. Thank you.

*If any further clarifications are required, please let me know.

siara-cc, 2023-03-16:
I have not really tried to use the lib concurrently.

As I understand it, esp32 has 2 cores and 1 is dedicated for WiFi communication and only 1 core is available for developer use?  Running single thread itself is an overload on ESP32 owing to limited memory and heavy SQLite library.  So this lib was not intended for multiple threads.  I think using multiple threads on the 1 core with limited amount of memory and heavy SQLite code is going to be an overload unless the threads are doing different things.

Zhu-jiatong, 2023-05-08:
The scenario I had in mind is when the SQLite database is used for session management:
- Using a ESP32 as a web server and [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer.git) as the library for the server.
- With each request for a resource on the server, the database has to be read.
- With each new login, a new session has to be created in the database (write).
In this case, when requests and logins are very frequent, I suspect that there might be near-simultaneous read and write action to the database, which would perhaps corrupt it or fail to complete the actions.