siara-cc/esp32_arduino_sqlite3_lib #85: Cannot use order by, limit and offset same time.

nhphuc699, 2024-01-24:
I use sqlite3 with the esp32-s3 board and microSD memory (4-bit 20MHz connection).

When I use the query: "SELECT p.*, nv3.MaSo AS MaSoNhanVienBanKiem, nv3.Ten AS NhanVienBanKiemName, nv2.MaSo AS MaSoNhanVienPhucVu, nv2.Ten AS NhanVienPhucVuName, l.NgayNguyenLieu, s.Ten AS SizeName, tp.Ten AS ThanhPhamName, nv.MaSo, nv.Ten AS NhanVienName FROM PhieuCan p LEFT JOIN Lo l ON p.MaLo = l.Id LEFT JOIN Size s ON p.MaSize = s.Id LEFT JOIN ThanhPham tp ON p.MaThanhPham = tp.Id LEFT JOIN NhanVien nv ON p.MaNhanVien = nv.Id LEFT JOIN NhanVien nv2 ON p.MaNhanVienPhucVu = nv2.Id LEFT JOIN NhanVien nv3 ON p.MaNhanVienBanKiem = nv3.Id WHERE p.Ngay = '20240124' order by p.NgayGio DESC LIMIT 10 OFFSET 1000;" or "SELECT * from phieucan p where p.Ngay = '20240124' order by p.NgayGio desc LIMIT 10 OFFSET 10;," it causes the board to reset. However, it runs normally when I remove "order by p.NgayGio."

My PhieuCan table has 1500 rows of data. When I execute the command "SELECT * from phieucan p where p.Ngay = '20240124' desc LIMIT 1000;", the returned data is around 400KB, and everything is normal. But when I select all 1500 rows of data, the board resets.

The data I select is loaded into a jsondocument variable and declared in PSRAM with a memory allocation of 3MB.

Can someone help me check if this is a limitation of sqlite3? Or is there any solution?

P.S. The above queries work normally when I connect to a computer using DB Browser for SQLite. Thank you very much.

siara-cc, 2024-01-25:
I think creating an index on PhieuCan.NgayGio could solve the problem.  If it does not, making this changes might solve the problem: https://github.com/siara-cc/esp32-idf-sqlite3/issues/22#issuecomment-1855778413

Order by would require lots of memory for sorting and it would use the index instead of sorting if the conditions are right.

nhphuc699, 2024-01-25:
Thank you. I will try to create an index on PhieuCan.NgayGio.
Additional information: I have tried to follow the method in the link you sent, but it still doesn't work.

nhphuc699, 2024-01-25:
Hi @siara-cc , I tried creating an index, and it worked well with limit, offset, and order by, but it only performs optimally with 1500 rows of data. When the data reaches 6000 rows per day grouped by the condition phieucan.ngay ='20240125' (6000 rows per day is also the maximum condition in practical use), order by cannot be used anymore. Currently, I almost have no solution for this issue and am forced to choose the option of limit and offset without order by. However, in the option without order by, I have to retrieve only 500 rows of data each time because if I retrieve more, the board will be reset.

I wonder if, based on your experience, there is any solution to my problem. Thank you for your feedback.

siara-cc, 2024-01-26:
There are a few things you could try:
- Try to get Sqlite query processor to use the index by checking EXPLAIN PLAN for different queries and making sure it uses the index to avoid sorting for ORDER BY.  Using Sub-queries instead of Joins could help.
- Try to use a board having more PSRAM to see if your sketch uses it at all.
- Store the data such that Sqlite won't have to retrieve large amount of data and filter and sort.  For instance you might have separate table for each day or month.  If Sqlite can't use index, it would have to retrieve all data into memory for filtering and sorting which is why the board resets.

nhphuc699, 2024-03-07:
Thank you @siara-cc , finally I have a solution to my problem. As you instructed, I have divided the data by day and created a new table for each new day. The queries to retrieve data do not perform joins or order by. If it is necessary to retrieve large amounts of data, I do it in a paginated manner, fetching only about 150 rows at a time (without using order by).

It may be due to my use of memory that data retrieval is not good for large amounts of data (6000 rows), such as operations to retrieve all 6000 rows, or using order by even though I have used an index

closed 2024-03-07 by nhphuc699