siara-cc/esp32_arduino_sqlite3_lib #86: Questions Regarding PAGE_SIZE and SQLITE_ESP32VFS_BUFFERSZ

slowpoison4, 2024-02-12:
Hi 
This is more of a Question/Observation. I have a very large program on ESP32 which uses the SQlite3 Library. Very early on i ran into the dreaded Sql %s error "out of memory". I dug through the code (Im a novice so not really sure about a lot of the implementation), But i saw  SQLITE_ESP32VFS_BUFFERSZ  defined as 8196. I decided to reduce this to 64 (Aggressive) but i saw a marked improvement in RAM usage. I wasnt getting the error any more and my code works fine. 

For reference, My code structure has 2 tables with 5 columns each storing texts. each column field size is about 192 characters. 

I also reduced the page size to 64 and once again saw major improvement in RAM utilization. I never faced issues with adding data, but deleting always failed before this. Now i can store 250+ messages on LittleFS with virtually no Performance impact. 

Am i doing something wrong here? I dont get why these changes worked because im dont know the importance of these values. Could you please shed some light on the same?

Once again, Big FAN of this library and really appreciate the work gone into this

siara-cc, 2024-02-12:
Hi @slowpoison4 
Thanks for the feedback!  About your queries:
- SQLITE_ESP32VFS_BUFFERSZ is just a buffer before making a write to disk so yes reducing it could make it occupy lesser memory.  I thought 8192 is not too much for esp32 but looks like it is worthwhile to reduce it.
- The minimum value for SQLITE_DEFAULT_PAGE_SIZE is 512 and not sure how it works for 64. It probably uses the higher value of sector size and this value.  I reduced the default page size to 512 to improve memory consumption. Probably you have a copy that is before this change.
Thanks for sharing your experience!

slowpoison4, 2024-02-13:
Hi @siara-cc 

Thanks for the quick response. I tried the Default Page size with the latest release as well, and I observed the same effect as before. The stability is fine as long as i use a Mutex to ensure no other task accesses/opens the db before the previous request is complete. I cannot open the db on my laptop with dB browser. It always gives me a corrupt file error (I suspect the page size being the cause here). But since i dont intend to use it in my laptop and purely using it for offline storage and retrieval it works perfect for me. 

Thanks for your time !

closed 2024-02-13 by slowpoison4