siara-cc/esp32_arduino_sqlite3_lib #90: Foreign keys and ON DELETE CASCADE not working properly

tesa-klebeband, 2024-05-09:
I am working on a project that utilizes foreign keys and their ON DELETE CASCADE property in order to remove unneeded entries from different tables. However, this function doesn't seem to work properly on my side as if I delete the row and thus also the column that the foreign key points to nothing changes and the row containing the foreign key still exists. I wrote a small snipped of code to test this behavior on another ESP32.
```cpp
#include <Arduino.h>
#include <sqlite3.h>
#include <SPIFFS.h>
#include <FS.h>

void setup() {
  Serial.begin(115200);

  sqlite3 *db;

  if (!SPIFFS.begin(true)) {
    Serial.println("An Error has occurred while mounting SPIFFS");
    return;
  }

  SPIFFS.remove("/test.db");

  sqlite3_initialize();

  sqlite3_open("/spiffs/test.db", &db);

  if (sqlite3_exec(db, "PRAGMA foreign_keys = ON;", 0, 0, 0) != SQLITE_OK) {
    Serial.printf("An error occurred while enabling foreign keys: %s\n", sqlite3_errmsg(db));
    while(1);
  }
  
  if (sqlite3_exec(db, "CREATE TABLE test (id INTEGER PRIMARY KEY, name TEXT);", 0, 0, 0) != SQLITE_OK) {
    Serial.printf("An error occurred while creating the table: %s\n", sqlite3_errmsg(db));
    while(1);
  }

  if (sqlite3_exec(db, "CREATE TABLE ref (test_id INTEGER, some_value INTEGER, FOREIGN KEY (test_id) REFERENCES test(id) ON DELETE CASCADE);", 0, 0, 0) != SQLITE_OK) {
    Serial.printf("An error occurred while creating the table: %s\n", sqlite3_errmsg(db));
    while(1);
  }

  if (sqlite3_exec(db, "INSERT INTO test (name) VALUES ('test');", 0, 0, 0) != SQLITE_OK) {
    Serial.printf("An error occurred while inserting data: %s\n", sqlite3_errmsg(db));
    while(1);
  }

  if (sqlite3_exec(db, "INSERT INTO ref (test_id, some_value) VALUES (1, 123);", 0, 0, 0) != SQLITE_OK) {
    Serial.printf("An error occurred while inserting data: %s\n", sqlite3_errmsg(db));
    while(1);
  }

  sqlite3_stmt *stmt;
  if (sqlite3_prepare_v2(db, "SELECT * FROM ref;", -1, &stmt, 0) != SQLITE_OK) {
    Serial.printf("An error occurred while preparing the statement: %s\n", sqlite3_errmsg(db));
    while(1);
  }
  while (sqlite3_step(stmt) == SQLITE_ROW) {
    Serial.printf("test_id: %d, some_value: %d\n", sqlite3_column_int(stmt, 0), sqlite3_column_int(stmt, 1));
  }
  sqlite3_finalize(stmt);

  if (sqlite3_exec(db, "DELETE FROM test WHERE id = 1;", 0, 0, 0) != SQLITE_OK) {
    Serial.printf("An error occurred while deleting data: %s\n", sqlite3_errmsg(db));
    while(1);
  }

  if (sqlite3_prepare_v2(db, "SELECT * FROM ref;", -1, &stmt, 0) != SQLITE_OK) {
    Serial.printf("An error occurred while preparing the statement: %s\n", sqlite3_errmsg(db));
    while(1);
  }
  while (sqlite3_step(stmt) == SQLITE_ROW) {
    Serial.printf("test_id: %d, some_value: %d\n", sqlite3_column_int(stmt, 0), sqlite3_column_int(stmt, 1));
  }
  sqlite3_finalize(stmt);
}

void loop() {
}
```
I am expecting this code to only output the rows one time when no delete statement has been executed, but i am getting the same row after the delete statement too. It would be great if this issue was eliminated.
Thanks

Zhu-jiatong, 2024-06-07:
Out of curiosity, has it been resolved?

tesa-klebeband, 2024-06-07:
No, still waiting for a response from the author.

siara-cc, 2024-06-08:
I was able to reproduce this issue using the console sketch, but so far unable to find the reason for this behavior.

Zhu-jiatong, 2024-06-10:
As shown in the [_Official Documentation_](https://www.sqlite.org/foreignkeys.html#fk_enable), the result returned from `PRAGMA foreign_keys` would indicate if the library is compiled with foreign key support enabled. 

I did a test using the aforementioned SQL query. No data is returned from the operation, indicating that the library is compiled with [SQLITE_OMIT_FOREIGN_KEY](https://www.sqlite.org/compile.html#omit_foreign_key) or [SQLITE_OMIT_TRIGGER](https://www.sqlite.org/compile.html#omit_trigger) defined.

siara-cc, 2024-06-12:
@Zhu-jiatong You are right. I thought since the OP turned on foreign_keys it does not matter if the library is compiled with OMIT_FOREIGN_KEY.
However, the library is not compiled with OMIT_FOREIGN_KEY on but only DEFAULT_FOREIGN_KEYS is set to 0. I will test with DEFAULT_FOREIGN_KEYS on but not sure if it will help.

siara-cc, 2024-06-12:
@Zhu-jiatong  @tesa-klebeband I enabled Foreign keys in config_ext.h and it is working fine now!

tesa-klebeband, 2024-06-12:
Thank you so much for enabling it by default! It now works without any problems. I am going to close this issue now.

closed 2024-06-12 by tesa-klebeband