siara-cc/esp32_arduino_sqlite3_lib #100: Cannot execute queries with JOIN operator

Bas1950, 2025-02-02:
First of all thanks for the excellent work!
Now my problem is that I cannot get a query executing that has a JOIN. All simple queries run fine!

Example:
when I run Q1:
"SELECT AF.ROWID,AF.NAME FROM AIRFIELDS AF WHERE AF.ROWID = 30" 
it calls the callback and returns the row

when I run Q2:
"SELECT AF.ROWID,AF.NAME FROM AIRFIELDS AF INNER JOIN AF_DIST AFD ON AF.ROWID = AFD.AF_ROWID WHERE AF.ROWID = 30"  
it does not call the callback.

BTW Q2 runsfine in SQLiteStudio (3.4.13)

Please help

siara-cc, 2025-02-03:
Hi, Only reason I can think of is that the result set is quite large, even then it I think it should call back.

Can you try running the query on the sqlite_console example which is based on `sqlite3_step` instead of callback?

siara-cc, 2025-02-03:
@Bas1950 You might load the sqlite3_console example given with this lib and run different queries on different tables to figure out what the problem could be.