siara-cc/esp32_arduino_sqlite3_lib #101: Disk I/O error when creating a table on SPIFFS with 24 columns

van-deste, 2025-02-17:
Hello,

In a project that I am developing, I have a table with 24 columns. In one of the ESP32 I have a SD card and all the operations work well. On other ESP32, I am using SPIFFS and I get this error when creating the table.

There is some type of limit by utilizing SPIFFS?

Thank you in advance.

[image]

van-deste, 2025-02-24:
Well, I did not know that...
I did not try to use LitleFS... is it simple to implement?

Yes my partition has space:
When creating the tables I do the follow:

`Serial.printf("Total space: %d bytes\n", SPIFFS.totalBytes());
Serial.printf("Used space: %d bytes\n", SPIFFS.usedBytes());`

And my serial monitor outputs the follow: 
Total space: 3848081 bytes
Used space: 6502 bytes

HankLloydRight, 2025-02-24:
> Well, I did not know that...
> I did not try to use LitleFS... is it simple to implement?

What IDE are you using?  I'd just goggle or AI help on how to upgrade.  It's almost plug-and-play but your code will need some minor changes.

van-deste, 2025-02-24:
I am using CLion IDE using the PlatformIO framework.
I will do the changes to implement LittleFS instead of SPIFFS and see if it resolves the problem

HankLloydRight, 2025-02-24:
I'm not familiar with CLion IDE, but I am using VSC with PlatformIO, so you shouldn't have any issues.

You can always revert back to SPIFFS if you need to. 
Good luck!

van-deste, 2025-02-24:
It is just a preference. I used to work with VSCode and PlatformIO, but I find CLion better for me. It is almost the same.
I will update you when I finish doing the changes.
Thank you for the fast response and your help.

van-deste, 2025-02-24:
@HankLloydRight problem solved. Thank you so much for your help!

siara-cc, 2025-02-24:
@HankLloydRight Thanks!
Yeah there have been issues using this repo with SPIFFS
https://github.com/siara-cc/esp32_arduino_sqlite3_lib/issues/50

closed 2025-02-24 by van-deste