siara-cc/esp32_arduino_sqlite3_lib #102: How do I update SQLite?

georgevbsantiago, 2025-03-26:
Hello. First of all, I would like to thank you for developing this library.

Are there any plans to update the SQLite version (3.43.2 - released on 2023-10-10) used by the library to a more recent version (e.g.: 3.49.1 - released on 2025-02-18)?
Version 3.49 has received improvements to the query planner.

If there is no plan to update, can you tell me if it is necessary to make many changes to the SQLite code to update the library? How can we track these changes so that we can try to update the library ourselves?

Thank you very much for this excellent contribution

siara-cc, 2025-03-26:
@georgevbsantiago Thank you for using this library. There are few esp32 specific changes made to the SQLite source before including it to this repo. I don't remember them offhand. You might do a diff of `sqlite3.*` to find out.

I will upgrade the version soon.

georgevbsantiago, 2025-03-26:
Thank you very much.
I will try to locate the `sqlite3.c` and `sqlite3.h` files from version 3.43.2 in the official SQLite repository to perform the diff.